rusefi/hw_microRusEfi #3: remove VAG specific pinout from schematics

rusefi, 2019-05-17:
alternator low side makes no sense for miata which uses high side alternator

let's remove all VAG-specific names from kicad files and have recommendations in documentation

injector1 2 3 4 are probably maybe the only name to keep in kicad since these are used for injectors in 99% of the cases

960, 2019-05-17:
Why not move the alternator to one of the High-Side drivers instead?
I dont know if it works as low side on any cars?
I kept it like this to make it understandable at a lower level.
It will apply to more people. We understand it no matter what it says.
There are nothing VAG specific there?

rusefi, 2019-05-17:
@960 my point here is that people would be confused. It is not understandable as is in my opinion.

In my opinion its safe to call pins injectors but the rest is not equally universal. I suggest everything else is called low side or high/low or high side output without calling vvt or alternator or anything

mck1117, 2019-05-20:
In the schematic, we should name things by what they can _do_, not what they're _for_.

In user-consumable documentation, we can provide guidance about what each pins could be used for particular functions (this may also vary per vehicle - high side and low side fuel pump relays both exist!).

960, 2019-05-21:
I dont see the issue.
This is a wirein ecu. We want users to wire it according to the default pinout in the dokumentation.
If everyone has different pinout, answering questions from frustrated users will be a nightmare.
The internal schematic are normally used for troubleshooting. Translating the pinout to hs1, ls4 or something users dont understand make no sense. 
The default pinout will work in 99% of engine schenarios, and it must be better answering the 1 % than the 99%.
This will newer be a diy ecu, with its 0603's and 0,5 pitch

If you dont understand it, please change it. I wont.

960, 2019-05-21:
Just please dont change any more symbols like you did with the arrays, without keeping annotations or assigning footprints. This broke the pcb drawing, and made a lot of extra work for me.

960, 2019-05-29:
I have started to edit the namings as I said I would from the beginning.
There have never been anything VAG spesific about it, I used names for parts found in vehicles I know, like coil, alternator, wastegate etc. just to keep track of what would be needed for any engine scenario.

I dont care more for VAG than any other cars, I am more of a Volvo fan.

The namings will be the same as in the software.

rusefi, 2019-06-05:
looks to be done

closed 2019-06-05 by rusefi