rusefi/hw_microRusEfi #5: separate constant 12v and ignition key

rusefi, 2019-05-17:
let's separate constant +12v from ignition ket 12v - this case we can implement tle8888 sleep delay later

rusefi, 2019-05-17:
Poke @dron0gus

rusefi, 2019-05-17:
[image]

960, 2019-05-17:
Doing this will take away one pin, as this will be a low amp wire.
If we use one of the existing pins, we only have one pin with High amp 12v.

Please explain further what this function do, and what we need it for.

mck1117, 2019-05-20:
Without standby power to the stm32's rtc/backup sram (via backup regulator or coil cell battery), this isn't particularly useful.  So I think for this ECU we don't worry about it, and leave the wiring as is.

960, 2019-05-31:
I think we can close this?

It would take one connector pin, and do nothing of use.

closed 2019-05-31 by rusefi