rusefi/hw_microRusEfi #25: Molex 48 pin silkscreen on the back

rusefi, 2019-05-28:
Can we please add some silkcreen on the back of the PCB for the connector?

definitely need some pin number reference at least for some pins otherwise we are blind

maybe even call out GND and 12v. 5v also?

rusefi, 2019-05-28:
@960 what do you think?

960, 2019-05-28:
I think we need to make something like this
[image]

rusefi, 2019-05-28:
Both :) We will have picture in the documentation but still some silkscreen would help

960, 2019-05-28:
Can we make help files work in TS?

rusefi, 2019-05-28:
Not sure what you mean.

rusefi, 2019-05-29:
We have progress :)

rusefi, 2019-05-29:
[image]

rusefi, 2019-05-30:
This is good enough at the moment but in my opinion more could be added. Adding more silkscreen to the connector is a great cost/benefit ration in my opinion.

Lowering the priority but keeping open for now.

rusefi, 2019-07-06:
looks great!

closed 2019-07-06 by rusefi