rusefi/hw_microRusEfi #28: Two more status LEDs

rusefi, 2019-05-28:
If we have space it would be nice to have two more status LEDs

rusefi, 2019-05-28:
Frankenso has a diode in addition to LED and R on the low side outputs

would microRusEfi also need a similar diode on low side output  LED?

[image]

960, 2019-05-29:
There will be no leds or testpoints at anything that already goes to the connector.

jharvey, 2019-06-01:
I'm a fan of these LED's on the injectors and ignition pins. I think Ola has not had the joy of diagnosing a failing injector after a mechanic has probed the harness with a meter and damaged a harness side of the connector. It is very common, and seeing a blinking light which says the ECU is good is really helpful in detecting if the problem is the ECU or the harness side. I understand the desire to no open the enclosure, but I also get angry at OEM's who fail to provide me with diagnostics information. A key feature of open source is that you have access to the low level information and can preform diagnostics. The MTBF of these LED components is very high, so they are not failure sources. Our failure sources are far more likely to be failed diagnostics by a mechanic who is in an uncontrolled world. I would like to encourage these LED's, however at the end of the day it's your design so your choice. I'm not looking to get into a debate about KISS or design elegance. I'm just trying to show why most OEM product include similar diagnostics LED's. In our case, our lack of control over the mechanic is why I think we need them when most OEM's do not have them. After all they are inside the enclosure, and if you never have a problem you'll never see them. You'll only see them if your already angry thinking the ECU has failed. These LED's can help prevent people from getting angry at the ECU people.

960, 2019-06-05:
Leds are added, I think we can close?

rusefi, 2019-06-05:
schematics show 4 status LEDs

closed 2019-06-05 by rusefi