rusefi/hw_microRusEfi #30: test pads for CRANK and CAM

rusefi, 2019-05-28:
for VR crank we really need a way to hook up logic analyzer for troubleshooting. While we have tiny LED pads to solder to I would prefer a nice 2mm diameter test pad so that if needed a pin could be butt-soldered

ideally same style pad is needed on PA5 track. PA5 would be the default crank at least in the beginning (I know schematics says AnOut7 = PA6 as cam but I believe this would NOT be my first preference) ICU is still the primary mode for trigger input and ICU works on PA5 not PA6

rusefi, 2019-05-28:
Q:  Why/when do you need pads for trigger?

A: for trigger logic troubleshooting we sometimes need to record raw signal using logic analyzer. Cranking, loss of sync - all these special cases can use logic analyzer recording of raw signal.

960, 2019-05-29:
What difference will these make compared to the external in connector?

You also have six 0603 footprints at the same place to solder wires.

I have tested these inputs with all kinds of signals, both hall, mcu output, VR, 1,5V battery etc..
Everything work perfect.

For the PA5 line, you will just be able to measure the same as the MCU sees?

I can add pads, I just need to know if it's really needed,

rusefi, 2019-05-29:
Q: What difference will these make compared

A: I need processed digital crank signal not source analog VR signal

960, 2019-05-29:
I think I have misunderstood the question.

I will replace the led with a pad.

I still dont see why it's needed, but anyway.

Those are the MCU inputs, and I thougt you could do the exact same reading with the MCU.

960, 2019-05-29:
Added

rusefi, 2019-05-29:
@960 while we can read from MCU its not comfortable.

i do not understand your goals. you are making this board hard for me to use, i am getting afraid that i do not want to be first user.

sure i would enjoy this as a product, but i am not interested in artificial hardships developing for this platform

960, 2019-05-29:
I have added the pads, so please just be happy with it.

rusefi, 2019-05-30:
I see CRANK pad

I see "CAM" words but I am not sure what these are referring to, please help me find CAM pad.

[image]

rusefi, 2019-05-30:
with new schematics CAM signal could be measured on the connector it looks like

no protection on CAM now?

[image]

960, 2019-05-30:
[image]

960, 2019-05-30:
Diode added just in case

960, 2019-06-04:
Pads added, and can be closed

closed 2019-06-05 by rusefi