rusefi/hw_microRusEfi #31: Larger T5V1/2 buffer caps

mck1117, 2019-05-29:
Buffer caps are currently 100nF, minimum recommended by datasheet is 400nF.  We should use 1uF or 1uF+100nF.

960, 2019-05-29:
[image]

960, 2019-05-29:
Isnt that what I have?

mck1117, 2019-05-29:
[image]

I mean for the 5 volt tracker outputs.

mck1117, 2019-05-29:
(other supplies (5v main, 3.3v, 6v prereg) look fine)

960, 2019-05-29:
Ok, I just made this from the reference design.

[image]

960, 2019-05-30:
Is it enough with larger values, or do we need larger footprints?

rusefi, 2019-05-30:
@mck1117 ?

mck1117, 2019-06-04:
footprints are already 0603, closing as it isn't a board issue.

closed 2019-06-04 by mck1117