rusefi/hw_microRusEfi #36: U5 wants that 4th pin connected to GND

rusefi, 2019-05-29:
U5 the LM2937 schematic is missing pin4, which is connected to pin2. This causes the PCB footprint to miss the GND connection and decreases it's thermal characteristics. 

@jharvey suggests adding the pin 4 to the schematic.

960, 2019-05-30:
This one are solved.

Pad number 4, (now 2) are connected to GND

[image]

rusefi, 2019-05-30:
@960 was this published? do we expect schematics to show 4 pins now?

[image]

960, 2019-05-30:
The symbol dont have pin 4

Instead of editing the symbol to be custom, I changed pad number for footprint to 2 to get ground at the Thermal pad.

jharvey, 2019-05-31:
I agree, this one is solved.

closed 2019-05-31 by rusefi