rusefi/hw_microRusEfi #37: USB D+ and D- should not have stubs or via's

rusefi, 2019-05-29:
USB D+ and D- should not have stubs or via's. There is a paper on that some where around here. The USB is different than our other low frequency signals, and needs more attention for reliable USB operation. 

D16 likely doesn't do anything as it's layout. The sharp angles after a stub will make most RF pass right by this. Most of the damaging energy in a ESD surge is high frequency. There is a white paper, I think it was from Intel, similar to the below, but for 2.0 and high speed and full speed stuff. It had many of the same recomendations and very similar data to back the suggestions. 
https://www.intel.com/content/dam/www/public/us/en/documents/white-papers/usb3-frequency-interference-papers.pdf

960, 2019-05-29:
This are taken care of now. please close.

rusefi, 2019-05-29:
@jharvey please review

rusefi, 2019-05-30:
I believe we have a via on D+/D-?

Need to find a way for @jharver or @mck1117 or @dron0gus to comment. I know _nothing_ about electricity.

[image]

960, 2019-05-30:
Vias in the ends are ok.

We have no way avoiding that to get thru pcb anyway.

The stubs are as good as they can get, Vias have enough space, and the traces have enough gnd plane all along next to them.

It's also routed to not go under the crystal

[image]

mck1117, 2019-05-30:
Those stubs aren't going to get much better, and usb FS (12mbit/s) is pretty tolerant of stuff like that anyway.  If it doesn't work, they're on external layers so we can cut them with a razor and remove in a future rev.

jharvey, 2019-05-31:
I think this will work just fine for our needs. If note I see the traces which might create issues can be scratched off. So we have a back up plan if we have a problem with this first spin. I think the USB is as good as it's going to get. I'm kind of tempted to suggest a 1uF cap but have not been able to check it against https://www.usb.org/sites/default/files/power_delivery_motherboards.pdf I suspect that is not a critical thing. I'm tempted to suggest just add some DNP pads for a cap, and we can deal with it later.

rusefi, 2019-05-31:
@jharvey 1uF cap where exactly? :)

jharvey, 2019-06-01:
per https://www.usb.org/sites/default/files/power_delivery_motherboards.pdf
2.2 shows 470pF parrell with 220uF. 
2.3.1 power lines must be bypassed with no less than 120µF capacitance of low-ESR capacitance per USB port
2.3.3 power and ground pins should be at least 0.050-in.-wide, with 1.5-oz. to 2-oz. copper
2.4.1 I do not like PPTC's for this. Most can be thermally activated in our normal operation parameters. If a PTC ends up being used, pay special attention to the devices temperature characteristics. I would simply negate this protection. Most upstream sources have good enough protection included. 

I think the above spec is very much on the safe side. I think we can negate the fusing concerns, we don't care if we take the bus voltage down. I rarely see 120uF caps on the power as they suggest. The small cap is for noise suppression of the arcing that happens when you connect the USB plug, so that 470pF is a good idea. We should also have pads for a 1uF or as they suggest 120uF cap. I'd just put what ever is "reasonable" on an 0805 for this cap. That might be a 1uF, or a 10uF, if we can get a 120uF, might as well follow the spec. I think the spec is excessively conservative with this cap. 

D17 will drop about 1V, so your 5V will really be more like 4V. As well if you draw 0.5A, that's 0.5W that the LED has to survive, with a poor heat sink. It is common to use a p-MOSFET, as seen below. Note the "Vin" is typically connected to GND via 10k. That circuit has an option which can be used to disable the 5V. We do not need that T2 MOSFET which is used to disable the 5V from USB. We just need the p-MOSFET and a high ohms resistor to GND. If we do this then we will have 5V instead of 4V when connected to the USB. 
https://forum.arduino.cc/index.php?topic=158717.0

The chip shown above can be found here. Other p-MOSFET's could be use if so desired. 
https://octopart.com/fdn340p-on+semiconductor-84326388?r=sp&s=Z0-2bL1_T3ieiurR2iQ5PA

More about reverse polarity protection found here http://www.youtube.com/watch?feature=player_embedded&v=IrB-FPcv1Dc As noted by D1001 on Frankenso, I'm also a fan of the clamping diode. 

I suggest PWR LED on USB. AKA not all USB cables include the power wires. It is common the cable only provides data. I know the concerns of these LED's, but they are also very common. 

At the end of the day, I think the USB as it is drawn now will mostly function, and will probably function well enough that no one cares. However there are likely some areas which can be made better. So I tend to flip flop on this. Some times I want the above noted components, just because an app note says you should have them. However other times I tend to think we don't need them, as they are just extra parts that don't really do anything we really care about. I don't think this CAP stuff is really very critical. I vote that this be closed, and what ever Ola decides to put in, should be good enough. I've shared some data so he can make an better informed decision.

closed 2019-06-01 by rusefi