rusefi/hw_microRusEfi #38: crank analog input needs pull-up

rusefi, 2019-05-29:
crank analog input needs pull-up resistor option

it's my opinion that we have space for all inputs to have both pull-ups and pull-downs not just 6 channels, it's my opinion that by not having pull-ups you limit the board too much

960, 2019-05-29:
Please elaborate.

Do you mean CAM, not Crank?

If a pull up are needed to read a Hall-Sensor I will add it, I did not know that.
As the ones I know have a positive(rising) signal, I have used a pull-down for the cam.

4 dedicated AN-TEMP Inputs
9 dedicated AN-VOLT Inputs
1 dedicated CAM Input
1 dedicated Crank Input

This is way way more compared to any other ecu's at this level, and exactly the same(With Pullups) as my $1800,- LINK ECU.

Are this really a important issue and a limitation???

rusefi, 2019-05-29:
miata na and nb generations both crank and cam are hall sensors are open collector outputs so you need a pull up to translate into digital form

@960 please elaborate what vehicles do you usually work on? could it be that you are making a VAG-specific board not universal board? you seem to not care about stepper idle and overall it looks like you have a pretty sptecific vehicle in mind

960, 2019-05-29:
I work with all kinds of european cars.

You wrote Crank in the issue?

I have checked the TLE8888 datasheet, and I have covered each and every scenario.

When you make such a claim I make anything at all of this to be VAG spesific, I expect you to explain exactly what you mean are VAG related!
That comment really offends me.

I am making this as close to any other aftermarket on the market as possible, working for any engine. 

I did not know about the CAM Pullup needed.
We need to find a solution that works for all hall sensors. 
It might be I am wrong, and a pullup are the right one.
If both, we need to add a switch IC that toggle between pull-up and pull-down.

960, 2019-05-29:
I was wrong, and a hall sensor should have pull-up.

Pull-up added.

960, 2019-05-29:
Can you also explain a bit better what you need for the Crank?

I dont understand what of this that wont work for your trigger.

[image]

rusefi, 2019-05-30:
@960 I am not aware of any issues with the current crank schematics.

I am only looking for a confirmation that this schematics (which I do not completely understand) would work fine on an open drain hall sensor.

960, 2019-05-30:
Yes, I can guarantee it will.

The cam sensor now have one pull-up, one series and a diode.

It's taken directly from a VAG oem schematic with open-drain hall

The Crank input have options for every possible scenario on planet.

closed 2019-05-30 by rusefi