rusefi/hw_microRusEfi #39: traces to not be under the screw heads, or under the enclosure

rusefi, 2019-05-29:
Can we limit to only two screws, and get some more clearance near the harness connector? That would allow traces to not be under the screw heads, or under the enclosure.

960, 2019-05-29:
I have rerouted the traces that got under the screw head when narrowing down edge cuts.

Under the pcb, there will be traces touching the enclosure all over it. 

For now, I would like to keep all 4 screws.

rusefi, 2019-05-29:
there will be plastic under the pcb

rusefi, 2019-05-30:
[image]

rusefi, 2019-05-30:
@960 says "The board doesn't touch the enclosure at top" - PCB is thinner and bolted to the bottom  leaving some small space on top everywhere.

closed 2019-05-30 by rusefi