rusefi/hw_microRusEfi #41: double-check TLE8888 outputs mapping

rusefi, 2019-05-30:
while we CAN configure TLE8888 with some flexibility, any weird mapping of parallel control pins to outputs would mean custom code or yet another layer of configuration for mapping

we have default initialization code which I assume maps inputs to outputs in some specific way. I would like to double-check that existing TLE8888 code plays nice with our choice of parallel control pins

rusefi, 2019-05-30:
@dron0gus @mck1117 
[image]
[image]

dron0gus, 2019-05-30:
4 x Inj and 4 x Ign are always mapped to IN_1..8.
Setup of IN_9..12  done through tle8888_config.direct_io array. Current default values:
	.direct_io = {
		[0] = {.port = NULL,	.pad = 0,	.output = 9},
		[1] = {.port = NULL,	.pad = 0,	.output = 10},
		[2] = {.port = NULL,	.pad = 0,	.output = 11},
		[3] = {.port = NULL,	.pad = 0,	.output = 12},
	},
So IN_9..12 will drive OUT_9..12. But there are no STM port & pin provided. So driver does not know how to drive these output is direct mode. You'll have to use native GPIOx_y instead of TLE8888_PIN_z...
Or someone should update config struct. For given screenshots it should look like:
	.direct_io = {
		[0] = {.port = GPIOE,	.pad = 10,	.output = 9},
		[1] = {.port = GPIOE,	.pad = 9,	.output = 10},
		[2] = {.port = GPIOE,	.pad = 8,	.output = 11},
		[3] = {.port = GPIOE,	.pad = 7,	.output = 12},
	},
Still not sure about outputs that needs to be driven throught these IN_9..12

dron0gus, 2019-05-30:
Updated direct io config for given scematic
	.direct_io = {
		/* IN9  */	[0] = {.port = GPIOE,	.pad = 10,	.output = 5},	/* 4.5A - VVT-2 */
		/* IN10 */	[1] = {.port = GPIOE,	.pad = 9,	.output = 6},	/* 4.5A - VVT-1 */
		/* IN11 */	[2] = {.port = GPIOE,	.pad = 8,	.output = 21},	/* 0.6A - WASTGATE */	
		/* IN12 */	[3] = {.port = GPIOE,	.pad = 7,	.output = 23},	/* 0.6A - RPM-OUT */
	},

rusefi, 2019-05-30:
High side mode requires more firmware changes - see https://github.com/rusefi/rusefi/issues/810

closed 2019-05-30 by rusefi