rusefi/hw_microRusEfi #49: BOM: J2 mouser part number

rusefi, 2019-06-05:
PinSocket_2x05_P2.54mm_Vertical_SMD needs part number

by the way J1 part number is Molex 105133-0031

U3 part number is Molex 500762-0481

ZHoob2004, 2019-06-10:
I propose Molex 15-91-0100

(also available as Molex 15-91-2100 with tin plating if you're into that sort of thing)

rusefi, 2019-06-16:
@ZHoob2004  15-91-0100 would not work

 15-91-0100 is a male part and st-link is male. We would need a female part on the PCB.

It would also be nice for the J2 female to have the key as pictured on brain board which uses through hole
[image]

rusefi, 2019-06-16:
@ZHoob2004 brings up a great point about stain relief. In light of SMD mount we  in fact should go with male header, let's just find a part with the key.

ZHoob2004, 2019-06-16:
Samtec HTSS-105-01-T-DV is a similar part with a plastic shroud that enforces connector direction, though it costs almost twice as much as the previously mentioned part.

rusefi, 2019-06-16:
@

ZHoob2004, 2019-06-17:
The shrouded part is too large for the existing footprints (would stick off the edges approx. 1.5mm on each side.)

I think we're stuck with the open version I suggested above, at least until there is a change to the board layout. I imagine we can get by with connector orientation clearly marked on the silkscreen.

rusefi, 2019-06-29:
[image]

jharvey, 2019-07-05:
About the J2 connector. I've been 50% about the via's inside the pads. I understand the school of thought that it sucks up the solder which can result in a poor connection. But I also do not have any real empirical data which claims it's bad. I can not find any real guides which detail why it's bad, etc. As well I'm not crazy about spending money on every board to have this connector, when it's only used once by developers. So I purpose.... an adapter PCB with a spring connector. Such that you just have pads on the PCB and then you press the programming dongle against the PCB and program away. This reduces cost of the PCB's as well it removes the debates of the entire via-in-pad debates. I see the below is not very expensive. 

https://www.digikey.com/product-detail/en/mill-max-manufacturing-corp/838-22-010-10-001101/ED10885-ND/6573078

So is everyone on board with changing to this kind of spring connector? Any thoughts?

rusefi, 2019-07-05:
Suggested part number if male 2.00mm not 2.54mm and I do not understand the whole process. I do not like the spring loaded idea because for me this is a new skill I have to learn, I would prefer to stay ignorant. I do not understand where would the new hand which would be holding the connector in place be coming from. I am afraid when I think about average users required to use it.

we have 4 options I assume at least?

1) can we do though-hole and shave the part on the read and press it against some film?
2) SMD no via
3) SMD with via
4) spring-loaded

@mck1117 how unstable is current SMD no via in your opinion? did you use it with wires or pins at all?

mck1117, 2019-07-05:
I wouldn't recommend hitting the current setup with a hammer.  Other than that, it's fine.  I use it with a 10-pin IDC ribbon cable with a 20-pin connector for my legit ST-link v2 on the other end of the ribbon.  There's a lot of total pad area under there, so I'm not worried about peeling it off of the board.

Option 1. sounds like a bad idea.  That's labor, and possible shorts to ground.
Option 2-4. The spring load pins don't necessarily need any special pad on the board to work.

Here's my proposal.  Leave the pads as is (SMD, 2.54mm spacing), which supports the following scenarios:
- Development: solder on SMD male/female connector.  Connect stlink.
- End user: program once using pogo pin jig, and don't install a connector.  If a user wants to become a developer and debug on their board, it's easy to add the connector.  End-user firmware updates should be done with DFU over USB, so you really shouldn't need a connector here.

Even if we want to use the spring loaded pins, there's no cost to using the full size pads for the solderable header, so why not do it? Leaving full size pads leaves our options open.  If we want a header on every board, that works.  If not, that also works.

rusefi, 2019-07-05:
@mck1117 "as is" is vague in my opinion. "as is 0.1" is different from "as is 0.2 is master"

please spell out "smd with via" or "smd without via" or else?

mck1117, 2019-07-05:
SMD with or without via probably doesn't matter.  It's a relatively large pad, so the solder-sucking effect from a single via would probably be minimal.  By "as-is" I meant "SMD sans via", but adding vias wouldn't hurt.  Either a header or pogo pins would work the same.

mck1117, 2019-07-05:
Such a pogo jig might look like this:
[image]
All it needs is some alignment pins to line the board up, appropriate spring loaded connector(s) to make contact with spots on the board, and something to hold it down.  It's really that simple.

rusefi, 2019-07-05:
^^^ looks totally unrealistic to me but I will believe it once somebody ships it to me.

Sounds like current SMD stays for now. If we believe current 0.2 master with vias is solderable and not worse than 0.1 let's move on?

mck1117, 2019-07-05:
Yeah, SMD+vias is fine.

jharvey, 2019-07-05:
In e-mail, OLA had suggested using a BDM frame similar to what is noted here. https://www.google.com/search?q=bdm+frame&tbm=isch&source=univ&sa=X&ved=2ahUKEwjVjcKp0pzjAhWwAxAIHXUGAKQQsAR6BAgFEAE&biw=853&bih=374&dpr=2.25

jharvey, 2019-07-05:
Can we find a lower cost version of the below?
https://www.digikey.com/products/en?keywords=TC2050-IDC-ND

mck1117, 2019-07-05:
While that's a neat connector, why?

mck1117, 2019-07-05:
Also, using a BDM frame as an *intentional* development tool seems silly.  They're designed for reverse engineering and probing of pins where you don't have a connector because you didn't design/fab/etc the hardware in question.  We're not in that scenario.  Let's do the right thing and just use a normal header 😄

960, 2019-07-05:
I recommend keeping it as is now.

However, I would have switched PA13 and PA14.

I dont remember if I did, can you check?

PA13 should be pin 2, PA14 pin 4.

jharvey, 2019-07-05:
The inside a china video posted a bit ago, shows them using a frame that is very similar to the BDM frame for programming and validation of PCB's. I think the frame is more than just a developers tool. Those are commonly used for both MFG programming as well as development purposes. 

About the why.... I guess largely for OCD purposes. I know that there will be endless debates about shed color, I mean about the entire solder sucking via's. I also know that OEM do it this way, so I kind of want to do it if for no other reason, just to get a better understanding of the OEM process. As well if we do it this way, then we can avoid the solder sucking land mine debates which I'm sure will come.

960, 2019-07-05:
As I use it all the time, I have to say a universal BDM frame are a little painful to use.

There are no alignment, so the exact placing take some attention.
I also need to be very careful nothing moves.

The spring loaded pins also bend easy.

The sucking vias are more of a issue at tiny pads, and end up beside the component instead of under.

The one mentioned was at one of the 0603 arrays, that was moved a little by accident.

rusefi, 2019-07-12:
We now have some silkscreen for key

rusefi, 2019-08-10:
Let's continue the discussion at #88

closed 2019-08-10 by rusefi