rusefi/hw_microRusEfi #51: BOM: mouser parts for 0603 and 0402 passive stuff

rusefi, 2019-06-06:
even 0603 and 0402 deserve part numbers

rusefi, 2019-06-11:
667-EXB-V8V102JV is 5%, I believe @mck1117  says to use CAT16-1001F4LF ?
667-EXB-V8V152JV is 5%, I believe @mck1117  says to use CAY16-1501F4LF ?

We also care about r14, 23, 25, 27 are 2.7k pull up resistors for analog inputs. @mck1117 @ZHoob2004 did you guys agree on part number?

rusefi, 2019-06-11:
Also R139 and R141

jharvey, 2019-07-05:
BOM is now fully populated, I think this has been addressed. I think this can be closed.

rusefi, 2019-07-12:
@jharvey since 1k and 1.5K are part of voltage divider there is an opinion that we want 1% not 5% resistors

suggested 1% part numbers are above

rusefi, 2019-08-11:
RN10 is now CAY16-1501F4LF 1%
RN1 is now CAT16-1001F4LF 1%

closed 2019-08-11 by rusefi