rusefi/hw_microRusEfi #58: loading caps on the crystal

rusefi, 2019-06-14:
@mck1117 says "loading caps on the crystal should be 8pF instead of 20pF"

mck1117, 2019-06-20:
confirmed - 8pf caps function properly

jharvey, 2019-07-05:
mck1117, have you seen the note on page 5 of Frankenso? I have some notes and a reference note when we were trying to figure out this intermittent clock issue. See below snippet which is on the Frankenso schematic. 
<code>
http://www.crystek.com/documents/appnotes/Pierce-GateIntroduction.pdf
PCB per predictions with SaturnPCB has less then 3.5pF traces, 
TPIC pins assumed 5pF
ESR = 80ohms max
Rf = 2meg could be between 1meg and 10meg.
Cload should be 18pF per XTAL datasheet
Cload = ([Cin+C1][C2+Cout])/(Cin+C1+C2_Cout)+PCBstray
Cload = ([5+24][24+5])/(5+24+24+5)+3.5= 18.0pF
C1=C2=C166=C167 = 24pF
Rs = 1/(2piFC2) = 1/(2*pi*8MHz*24pF) = 829ohms, 820ohms is close enough = R177
</code>

I think we technically for this one, we want 4pF not 8pF. See my calcs from here
https://docs.google.com/spreadsheets/d/1O6-u6T_O083pT9SEm70DGKKy6YZJ89tGeXWsZTZnmM4/edit?usp=sharing

mck1117, 2019-07-05:
I hadn’t seen that yet, no. The boards I’ve assembled so far have 8pF caps, and seem to work fine, fwiw. Is there a way to empirically measure whether the crystal is over or under loaded? I have a 200mhz scope at my disposal, and a 4ch/2ghz one at work if I need it.

http://www.crystek.com/documents/appnotes/Pierce-GateIntroduction.pdf<https://eur02.safelinks.protection.outlook.com/?url=http%3A%2F%2Fwww.crystek.com%2Fdocuments%2Fappnotes%2FPierce-GateIntroduction.pdf&data=02%7C01%7C%7Cdb6776e5253c4bb56a3a08d700ef04d0%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C636978898883165381&sdata=9PK%2F6Arba6gVao2uFQKQc2q4ibz%2B2LCXXYDWl%2BKk5x0%3D&reserved=0>

https://docs.google.com/spreadsheets/d/1O6-u6T_O083pT9SEm70DGKKy6YZJ89tGeXWsZTZnmM4/edit?usp=sharing<https://eur02.safelinks.protection.outlook.com/?url=https%3A%2F%2Fdocs.google.com%2Fspreadsheets%2Fd%2F1O6-u6T_O083pT9SEm70DGKKy6YZJ89tGeXWsZTZnmM4%2Fedit%3Fusp%3Dsharing&data=02%7C01%7C%7Cdb6776e5253c4bb56a3a08d700ef04d0%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C636978898883175385&sdata=ufzmqtSzKPVVaTvOB%2FEOz2CjuGm%2BsUic%2FA6eFHGE0as%3D&reserved=0>

jharvey, 2019-07-05:
I don't know a way to measure if the loading is proper. I understand the problem is mostly a failure to start issue. If you change your pF's you can wander the 8MHz slightly, but I don't think we care about the slight warping. I guess we could measure it bey building 1k boards and see if they all start at the min and max temperatures. If yours is starting, with 8pF then it's probably fine. I see 4.7pF is available, so I updated the BOM and schematic to call for that.

closed 2019-07-07 by rusefi