rusefi/hw_microRusEfi #65: document power and thermal expectations

rusefi, 2019-06-23:
@mck1117 says
I hooked up current consumption monitoring, and linear regs are killing us lol
235mA
2.9 watts
frankenso pulls around 85mA, 1 watt

We probably should not care but we can document just to be nice about documenting

@jharvey FYI

jharvey, 2019-07-05:
Can I get more details? Including
-- how low can we go on volts 
-- peak amps with no load so no injectors connected
-- Inrush amps
-- Amps at 18V
-- Amps at 14.4V
-- Amps at 12.0V
-- Amps at 8V. 

While I'm asking for all sorts of stuff. how about amps while right side up and while upside down. then amps while facing north, and amps while facing south :)

jharvey, 2019-07-05:
I added "rusEFI MRE was measured to consume 235mA from the 12V battery, for 2.9 watts" to both the PCB as well as the schematic. This is handy as it give me an idea about how I should set my power supply amps limit for my bench top supply. AKA don't let the smoke out when there is a mild issue.

closed 2019-08-11 by rusefi