rusefi/hw_microRusEfi #69: LD1117 is a dollar cheaper

rusefi, 2019-06-28:
LM2937 is $1+ while LD1117S33TR from mouser is $20@100. Shall we move to LD1117S33TR?

jharvey, 2019-07-05:
There is qty 1 of this component, I say use OEM official parts, we can deal with making it slightly cheaper once we know it works well. It's cheap insurance for a flat $1 fee. I suggest this is closed.

abelom, 2019-07-05:
I do not follow. Why is LD1117S33TR from mouser not OEM official part? Why do we not trust LD1117S33TR from mouser? @jharvey

I agree this is low priority

DonaldBecker, 2019-09-17:
The motivation to change from the LM2937 should be because it is a specialty part.

The LM2937 has the unusual feature that the tab is ground.  This is convenient for mounting to a heatsink, but otherwise is a disadvantage. Most regulators have a pinout that puts the input or output on the tab, which allows mounting the die directly on the tab for best thermal and electrical conductivity.

This electrical connection to the tab is associated with the regulator's internal construction and characteristics.  The classic high-drop-out LM317 has the tab as an ouput, with a NPN Darlington pass transistor mounted directly to the tab.  A tab as an input pin suggests a PMOS pass transistor which allows a very low dropout voltage, low quiescent current, but a more limited input voltage range and capacitor ESR.

We don't use any of the special features of the LM2937 (ground tab, high voltage input, reverse voltage protection).  It's reasonable to switch to the standard tab-output pinout, which allows using a wide range of better performing, lower cost regulators.