rusefi/hw_microRusEfi #75: LEDs for crank and/or cam

rusefi, 2019-07-05:
We have test pads but it is my opinion that our custom base would appreciate LEDs on crank and/or cam inputs

mck1117, 2019-07-05:
Is there a software solution we can use instead? Surely we can speak the right incantations to Tunerstudio to get this to work:
[image]

abelom, 2019-07-05:
Noth solutions are valuable. We have the ticket for composite logger with no progress for a while :(

jharvey, 2019-07-06:
Was the LED's loading a concern on either CAM or CRANK? With a 1k LED resistor, I expect the LED to consume about 4mA.

rusefi, 2019-07-06:
probably cam hall since very simple schematics? probably not vr crank since fancy tle8888 is dealing with it?

jharvey, 2019-07-06:
For CAM I just added a series LED. I think I can do the CRANK but after the 8888 has decoded it, and is generating digital pulses. Hmmm, I have no idea if this will work on the 8888. I do not see an mA rating for VROUT. I do not know if that is sink vs source. So this may not work for the VROUT signal. However I have added some pads so we can try it out.

mck1117, 2019-07-06:
Added a series LED where? Can you update the schematic PDF, or attach a snip of the change here?

Also, don't add extra test pads to the board for things like this.  We made TLE8888 breakout boards for exactly this purpose, so let's test out feasibility on those before we spin boards without knowing whether it'll work.

rusefi, 2019-07-06:
@mck1117 just manually merged latest
@jharvey can you please generate fresh .pdf?
Jared shall we close this now or still things to change?

mck1117, 2019-07-06:
The cam LED, as currently wired, won't work.  Forward voltage for an LED can be upwards of 3.5v, so the highest voltage seen at the cam pin could be as low as 1.5 volts.  That's very very close to the STM32's input transition voltage, so it may not actually read edges on the cam input.

rusefi, 2019-07-07:
Would a diode help CAM LED @jharvey @mck1117 ? I am sorry I know it's not my place to talk EE :)

closed 2019-07-07 by rusefi