rusefi/hw_microRusEfi #81: LED symbol was not shared, and needs to be corrected.

jharvey, 2019-07-08:
LED symbol "Device:LED" was not shared by OLA, and has the same name as a KC standard library. This causes the wrong symbol to be used. This needs to be updated, perhaps using a symbol name that is not already a standard name. The current LED shows backwards, as pad 1 and pad 2 are backwards.

rusefi, 2019-07-08:
@jharvey would it be best to simply use "Device:LED" from KC standard library but fix the usages to be correct? this case we use the right default symbol as intended, it's not like we have custom R or C symbols?

jharvey, 2019-07-09:
Hmm, I'm trying to change the title. I can't seem to change it. The symbol was apparently never shared, so it was just plane wrong. I wanted to update the title to reflect this. 

I'll figure out how to correct it. Either change the footprint, or change the symbol. I'm not sure what the solution will be.

960, 2019-07-09:
How about just turning them around?
You can do that in eeschema, just rotate..

jharvey, 2019-07-09:
KC footprint was over-sized, and does not have a clear polarity indicator. So I updated the footprint to match the MFG's suggested pads, as well I gave it a clear polarity indicator.

closed 2019-07-09 by jharvey