rusefi/hw_microRusEfi #86: VDD BOOT0 usability

rusefi, 2019-07-13:
We still do not have https://github.com/rusefi/rusefi/issues/809 and I assume that even once we have it sometimes things would not go well, so

We need some usability for VDD<>BOOT0 jumping. At the moment we have R26 which I assume is the default 0603 which is not very useable for jumping while rebooting.

We need an SMD jumper header so that a jumper could be put to connect VDD with BOOT0 if desired

Abricosvw, 2019-07-28:
i think best approach is not exposed pns for jumper but a touch button (not toggle button)

rusefi, 2019-07-28:
@Abricosvw  Omron Electronics Inc-EMC Div B3U-1000P for $1.50 is too much money!

There must be something similar waaaay cheaper on eBay. Overall I like the push button idea!

rusefi, 2019-07-28:
I like https://www.ebay.ca/itm/100PCS-3-4-2mm-Mini-Tact-Tactile-Push-Button-Switch-SMD-2Pin/232305685293

jharvey, 2019-07-29:
What is the digikey equiv to add to the BOM?

rusefi, 2019-08-10:
I believe the button is a bit larger than the current footprint. Can we expand by 1mm?

See image

[image]

jharvey, 2019-08-11:
See if this one fits better. Also this is really close, did you verify you had 100% scale? It is common that printers will print 98% scale, which can bugger things like this. I often have to print at 102% scale to get closer to the 100% scale in the real world.

rusefi, 2019-08-11:
Looks good now!

closed 2019-08-11 by rusefi