rusefi/hw_microRusEfi #87: What are R44 & R46 used for?

rusefi, 2019-07-13:
We have 0R R44 and DNP R46, what is the use-case for those? Would we remove those once some testing is done?

rusefi, 2019-07-13:
@mck1117  says
r46 selects vref = 3.3v
r44 select vref = vcc * 0.6

rusefi, 2019-07-13:
When would we want 3.3v instead of vcc * 0.6?

rusefi, 2019-08-10:
Is anyone planning to even not have 0R R44 and DNP R46?

mck1117, 2019-08-10:
We probably actually want to use r46, and NOT r44.  The trouble is that if we use r44, vref is out of spec for the stm32 if the board is supplied 5v via USB, and not 12v via the connector.

The side effect is that battery voltage measurement is unpredictable when vref is very low, so we may be unable to detect a rising edge on the battery voltage if r44 is used.

My board and russian's board both use r46.

rusefi, 2019-08-11:
OK whatever low priority for 0.3 ;)

rusefi, 2019-08-19:
@mck1117 so let me flip the question :)

Is anyone planning to even not have 0R R46 and DNP R44? Shall we make R46 a trace and remove R44?

mck1117, 2019-08-19:
Is there risk to leaving both footprints?

mck1117, 2019-08-19:
(not that I'm necessarily advocating for leaving both -- routing a trace in place of r46 is probably the right thing to do)

rusefi, 2019-08-19:
This is truly a low priority, but we shall at least fix the BOM to suggest 0R R46 and DNP R44 for now

jharvey, 2020-04-05:
This is obsolete at this point in time. Suggest closing this issue.

closed 2020-04-05 by rusefi