rusefi/hw_microRusEfi #88: SWD silkscreen

rusefi, 2019-07-13:
Would it be possible to add silkscreen to 6 pins of SWD header?
SWO
(N)RST
(S)CK
(SWD)IO
GND
VDD

Both 5v and second GND/VDD are less important

rusefi, 2019-07-21:
We have used proper footprint on rusEfi brainboard (it's upside down comparing with MRE but note where GND is)

[image]

rusefi, 2019-07-21:
"SWD is always mapped on PA13 (SWDIO) and PA14 (SWCLK)"

there is also some problem with SCK net name PA14 should be clock and SCK is clock? the pin which is NOT IO not CLK not NRST should be PB3/SWD?

[image]
[image]

rusefi, 2019-07-21:
also are all IDC cables the same? if we plan to use IDC cable do we need to invert the pinout since the female/female cable flips it?

rusefi, 2019-07-27:
Looks like at the moment schematics has the correct names/functions for tracks/pads, while PCB has some errors related to PA14 not being CLOCK and PB3 not being SWO?

rusefi, 2019-07-27:
See also some comments at #49

rusefi, 2019-07-28:
@jharvey reminds us that PB3 is used both as SWO for singe wire debugger as well as SPI but clock used to SPI TLE8888

rusefi, 2019-08-10:
Here is a picture of an IDC cable. Both sides have the key on top, red and yellow wires are both in top row, there is continuity between the holes into which red and yellow jumpers are plugged into

[image]

mck1117, 2019-08-10:
ST now has the [STLINK-V3MINI](https://www.st.com/content/st_com/en/products/development-tools/hardware-development-tools/hardware-development-tools-for-stm32/stlink-v3mini.html).  This is the new official debug tool from ST, and **is under $10** (but isn't yet in stock anywhere - it will be soon).

It has a 14 pin 1.27mm connector on the bottom, and includes a 14-pin IDC cable.  It's compatible with the 10-pin Cortex debug header, but adds an extra pair of pins on each end.

I would recommend using a connector like this one: https://www.samtec.com/products/ftsh-105-01-l-dv-007-k

Since it has no ends on it, it can accept either the standard 10-pin cortex header, or the ST 14-pin extended connector on the STLINK-V3MINI.

mck1117, 2019-08-10:
In addition, that maintains backward compatibility with older st-links (and other programmers) by supporting the industry standard 10-pin Cortex debug header.

rusefi, 2019-08-10:
STLINK-V3MINI with 1.27mm is interesting for MRE 0.9 or MRE 1.0, for now let's survive the next 6 months with 2.54mm

It looks like IDC cable mirror pins left to right but keeps top and bottom? I.e. top-left pin becomes top-right pin here. All three female connectors the key is facing the camera

[image]

mck1117, 2019-08-10:
IDC cables don't mirror anything.  You can route the cable in the "tab side" or the "flat side" of the connector, but that **does not** change how the connector pins map to the wires in the cable.  Don't look at which direction the cable exits the connector, look at the tab location.  The red stripe on the cable goes at the pin 1 end, which is the only thing that matters.

mck1117, 2019-08-10:
[image]

jharvey, 2019-08-10:
I agree with mck1117 what I was not sure about is if the cables that rusefi had were IDC cables. They are low cost china, so perhaps crimped upside down, or backwards, or some other thing. As far as I can tell it looks like the cable should not be mirroring the connector.

960, 2019-08-11:
I use these cables for much of my tuning equipment, and they never mirror anything.

Twist the cable and put the connectors next to eachother the same way, and you'll see it.

The official cortex header are the right thing to use.

rusefi, 2019-08-11:
Yes the whole mirroring is weird

My "ebay black pink" cable does not match the picture above, I assume the cable is the problem
[image]

mck1117, 2019-08-11:
There's no mirroring.  Pin #1 is always in the same place.  The point of an IDC cable is that you can install multiple connectors on the same cable, all with the same pinout.  Every picture you've sent shows a properly assembled IDC cable.

960, 2019-08-11:
It's just like in old stationary computers. they was full of them.

From the mainboard to the harddrives you had several connectors for HDD's, DVD-drive etc.. at the same cable

rusefi, 2019-08-11:
[image]

jharvey, 2019-08-11:
Per Andrey's picture, we do seem to have a problem with the ribbon cable. This is because we are flipping, while ribbon cables typically sit flat. The question is what should we do about it? Should the PCB pin out be changed to require the ribbon cable?

jharvey, 2019-08-11:
In Andrey's picture, you can see RST is flipped from the right side of the connector, to the left side. You can see the red tracer is correct, but geometry has flipped the pin out, because the connector is flipped on the other end of the ribbon cable.

jharvey, 2019-08-11:
The ribbon is presenting concerns and potentail problems. We should probably just plan for the programming header directly, with silk screen to show the key position. With some luck and with some time, this will be obsoleted any how.

jharvey, 2019-08-11:
I vote to close this one.

closed 2019-08-11 by rusefi