rusefi/hw_microRusEfi #93: sorta kind of backfeeding? isolation?

rusefi, 2019-07-26:
why did MRE 0.1 reboot on me while cranking? Do we have anything preventing backfeeding? isolation?

rusefi, 2019-07-28:
@mck1117 says
"re: low voltage dropouts, I removed r21, and now my board works down to 4.9 volts
it used to die at 7.9"

rusefi, 2019-08-02:
@jharvey let's remove R21?

@mck1117 says " with that resistor removed, it survived a (logged) minimum voltage of 7.6 volts"

[image]

jharvey, 2019-08-02:
I think R21 is valid, it's just in the wrong place on the schematic. See R107 from the reference TBOARD schematic. Notice it's on the battery side not the key side of the filtering.

rusefi, 2019-08-02:
[image]

rusefi, 2019-08-11:
We will see how it goes with 0.3 I guess

closed 2019-08-11 by rusefi