rusefi/hw_microRusEfi #99: 5v/12v option is not consistent with default naming

rusefi, 2019-08-03:
I believe the intention was for coil drivers to be hard-wired to +5 and for additional high-side to have +5v or +12v flexibility

we then had a mess with which stm32 line is wired to what output

looks like we still have some effects of this mess

also "8 channel label" on top of the PDF page is wrong, we only have 6 channels

[image]

mck1117, 2019-08-03:
Yes - I think a few things on the schematic are still named wrong.  I know that the default config/pin names in the `mapping.yaml` match the wiring pinout doc, so any errors are constrained to the schematic.

rusefi, 2019-08-04:
@mck1117 I believe PCB silkscreen matches schematics thus NOT matching mapping and online pinout at https://github.com/rusefi/rusefi_documentation/wiki/Hardware_microRusEfi_wiring

Open question which is the source of truth I guess

mck1117, 2019-08-04:
Oh - no idea what the silkscreen says since my v0.1 doesn't have it.  The actual PCB wiring matches the pinout at that link, at least on v0.1.

rusefi, 2019-08-11:
@jharvey schematics still says "8 channel" which is not correct

@mck1117 schematics page still shows 5v/12v on IGN pins not HS pins. This could be where we would need to change the pinout on the wiki? I am a bit confused.

rusefi, 2019-08-11:
Considering R59/R57 is wired to U31, U31 has to the HS1/HS2

U32 and U33 should be the ones COIL1/2/3/4

So open question if we move R59/R57 or change traces or change the pinout. Opinions?

rusefi, 2019-08-11:
"8 channel" is now resolved

rusefi, 2019-08-11:
@mck1117 there will be mapping yaml change to match the internal pinout change I believe

closed 2019-08-11 by rusefi