rusefi/hw_microRusEfi #113: Power supply load dump protection

mck1117, 2019-09-05:
We currently have no protection against the load dump tests enumerated in ISO 16750-2 (namely the 80-100v positive pulse).

This one is pretty easy to resolve - just add a pair of back to back TVS diodes across the power input pins.

DonaldBecker, 2019-09-07:
Most alternators have load dump protection, putting the system in Test B.

My quick reading of the standard suggests that in Test B the device only needs to survive a 35V pulse.  Most of the work to do this is already designed into the TLE8888 with the specified pre-regulator.  For instance the unprotected inputs are rated to withstand 40V (which is 35V + margin).

Vehicles without load dump protection are antiques uhmm classics with period-correct parts.  For that narrow use case it's reasonable to expect the load dump protection to be external and sized to the unique requirements.  (If you want to use an original generator, ISO-16750 isn't your target.) 

It's worth noting that passing the test requires the device to be at Class C or above, which means continuing to work normally after the test.  That rules out simply blowing an external fuse with a TVS (which I read as Class D) or an internal fuse or the TVS itself (Class E).  This might be a case where being "extra safe" by adding protection to an already-OK system fails the test and decreases real-life reliability.

mck1117, 2019-09-07:
Most modern alternators have load dump protection - but we can't guarantee that you're running a modern alternator.  If we were building a setup for a particular vehicle, we could put ourselves in Test B, but there are people out there already considering putting one of these in vehicles that may be old enough to not have load dump protection built in to the alternator.  For example, @ZHoob2004 has a 1970s Datsun Z car that I doubt has it, and my 1991 Volvo is probably a tossup on whether it has it.

> For that narrow use case it's reasonable to expect the load dump protection to be external and sized to the unique requirements.

Remember - the target customer here isn't an engineer, they're some random car enthusiast who might pull a 50 year old alternator out from the back of a shelf and slap it on their car (then have the battery cable fall off while driving - causing a load dump scenario).

As for actually passing Test A, the SM15T33CAY is probably all we need across the power rails to do it.  And I think we would stay class C even if fused.  The datasheet provides example voltage/current at the device for the various positive/negative load dump tests, and the peak current on any of them is for about 5 amps, for only 500us, which probably wouldn't even blow a 1A fuse.

(By the way, the C in the part number above indicates that it's two back to back diodes, so it also protects against negative pulses)

Here's app note A2689 from ST that recommends the ST15T33CAY, with some analysis about how much TVS you need: https://www.st.com/resource/en/application_note/cd00181783.pdf

mck1117, 2019-09-07:
Plus - that TVS diode is only $0.80.