rusefi/hw_microRusEfi #119: test pads for TTL pins

rusefi, 2019-10-17:
Low-probability scenario: remote tuning via a data modem, data modems probably prefer TTL RS232, can we expose test pads for one of stm32 uart devices?

DonaldBecker, 2019-10-17:
In my layout I added a 2x5 pin SMD header that brought out pins that could be mapped.

5V    12V
B8     3.3V
C11   D0
A15   C10
GND  C12

These can be mapped to:
Pin   Some interesting connections
PB8  I2C1_SCL/CAN_RX1
PC11  UART4_RX/ SPI3_MISO/USART3_RX
PD0   CAN1_RX
PA15  SPI1_NSS/ SPI3_NSS
PC10  SPI3_SCK/ UART4_TX/ USART3_TX
PC12  UART5_TX/ SPI3_MOSI/ USART3_CK

My layout should be considered a *proposal* rather than a settled design.  These are exactly the decisions that end up being "it would have been so much more useful if you made this minor change". 

The intent was that this header could be used for a controlling a SPI device, adding serial communication over bluetooth or WiFi, or adding a SPI memory chip.  Note that the 8 of the pins directly map to a SPI flash pinout.

Obvious drawbacks:
- It *almost* but not quite supports dual serial ports, a second CAN channel, I2C, etc.
- It almost supports a SD card or SDIO.
- No analog input pins are provided.
- It could support I2S, but that seems useless here, unless someone uses it as a sound input for knock processing.

Note that the pads are for a 0.1" spaced header, which takes a bunch of room.  A 0.05" / 1.27mm header would allow more pins or a smaller footprint.  The header should be female to avoid pins being exposed to accidental contact, especially as 12V is on a vulnerable outside edge.

DonaldBecker, 2019-12-22:
For 0.4.6 I've updated the communication header pinout to use PB9 instead of PD0.
This allows it to support I2C or CAN (I2C Ch2 or CAN Ch1)

5V     12V
PB8   3.3V
PC11 PB9
PA15 PC10
GND PC12

Pin  Pin assignment options
PB8 I2C1_SCL/CAN1_RX
PC11 SPI3_MISO / UART4_RX/ USART3_RX
PB9 I2C1_SDA/CAN1_TX
PA15 SPI3_NSS
PC10 SPI3_SCK/ UART4_TX/ USART3_TX
PC12 SPI3_MOSI/UART5_TX/ USART3_CK

jharvey, 2020-04-05:
This is obsolete, suggest closing this issue

closed 2020-04-05 by rusefi