rusefi/hw_microRusEfi #121: Why are we using nonstandard debug header?

mck1117, 2019-11-08:
There exists a standard debug header that is:
- physically smaller
- industry standard (no more eBay pinout mysteries and fake programmers)
- keyed (impossible to plug in backwards)

Here's the doc on the Cortex debug headers.  We'd be interested in the "Cortex Debug Connector (10-pin)" described on page 2.
http://infocenter.arm.com/help/topic/com.arm.doc.faqs/attached/13634/cortex_debug_connectors.pdf

When the ST-LINK/V3 comes out soon, we'll be able to buy just the $11 programmer:
https://www.digikey.com/product-detail/en/stmicroelectronics/STLINK-V3MINI/497-19530-ND/10266291

Today, the connector can be used with an official ST-LINK/V2:
https://www.digikey.com/product-detail/en/stmicroelectronics/ST-LINK-V2/497-10484-ND/2214535
Plus an adapter to the 1.27mm 10-pin:
https://www.digikey.com/product-detail/en/olimex-ltd/ARM-JTAG-20-10/1188-1016-ND/3471401

mck1117, 2019-11-08:
update: Mouser has the stlink-v3 in stock!
https://www.mouser.com/ProductDetail/STMicroelectronics/STLINK-V3MINI?qs=mAH9sUMRCtvrXm%2FF%2FC0CYg%3D%3D

mck1117, 2019-11-08:
And here's an example connector we could put on the board:
https://www.digikey.com/product-detail/en/samtec-inc/FTSH-105-01-F-DV-007-K/SAM15640-ND/6693339
It's quite small, only requires around 6x7mm of board space.

DonaldBecker, 2019-11-09:
The argument for the current connector is:
   A USB dongle ST-Link v2 is $2
   The connector is inexpensive, at around $0.25
   The connector pins are enclosed, making it safe for poking around.
   Both 5V and 3.3V can be supplied through the connector, allowing a single debugging cable

The argument against is
  The footprint is huge
  Using a female header requires the ST-Link dongle to plug directly into the board
  Doubled pins are certainly not needed.

Alternative are
   Omitting programming header completely
       DFU for installation and updates is good enough for end users
   TagConnect footprint
        Tiny footprint, zero cost
         Expensive commercial cable or fiddly home built cable needed.
  Changing the suggested header to a male pins
       Inexpensive, and allows using cheap cables to the programmer
       Still the same footprint size, and exposed 5V pins sticking up is risky
   Changing the header to shrouded male pins
       Keyed ribbon cable to avoid reverse connection, mitigates exposed pins, inexpensive
       Effectively even bigger footprint (no flat components allowed in surrounding area)
   3 or 4 pin SWD header in the Discovery board pattern
         Inexpensive, smaller, does everything
   Standard JTAG, 2.54 or 1.27 spacing
          More pins, none of which add value
   ST-Link v3 header
          Programmer is only about $10
          Expensive header $3.20 ea
          Very little benefit for extra pins
          Very new and still uncommon

Right now I'm leaning to a 3 or 4 pin SWD only connector, with a leg-less TagConnect footprint.
Perhaps with a 2x2 SMD pin header rather than the 1x4 Discovery header for space savings
That provides almost complete debugging at minimal cost, usable with the existing ST-Link debuggers.

jharvey, 2020-04-05:
This is obsolete, suggest closing this issue. We appear to have both standard and less standard programming headers now.

mck1117, 2020-04-05:
We have 10 pin cortex now?