rusefi/hw_microRusEfi #134: Still an open issue if we even need BAS70-04

rusefi, 2020-02-24:
43;"D10,D26,D25,D24,D18,D12,D11,D9,D8,D7,D6,D5,D4,D3";"SOT-23";14;"BAS70-04";;;

was it replaced by 

16;"D39,D13,D38,D40,D16";"SOT-23-6";5;"SRV05-4";;;
?

rusefi, 2020-02-24:
@DonaldBecker @mck1117 @anreika-git @dron-gus @jharvey opinions?

rusefi, 2020-02-24:
If we still node those shall we migrate to smaller footprint PN?

sot323 - also called sc-70

https://lcsc.com/product-detail/Schottky-Barrier-Diodes-SBD_DIODES_BAS70W-04-7-F_BAS70W-04-7-F_C134389.html
or
https://lcsc.com/product-detail/Schottky-Barrier-Diodes-SBD_DIODES_BAS70W-04-7-F_BAS70W-04-7-F_C134389.html

mck1117, 2020-02-24:
here's how to test it:
1. supply normal power to board
2. connect +12v to analog input, measure voltage on the ECU-side of the 10k dropper resistor:
[image]
(apply 12v to left arrow, measure at right arrow)
3. If the voltage is below ~6 volts, then the diodes work great.  If the diodes or opamps release smoke, the diodes do not work great.

rusefi, 2020-02-25:
@DonaldBecker says "I was unable to kill the board without the bottom layer protection installed.
Using the 3peak opamps, with higher allowed injection current, should mean that we might not even need the srv05 diodes on the top layer, but those are not large or expensive.

Injection current test was at 15.0V

With is a 10.0V drop across the input diodes to the 5.0V opamp power rail.

Presumably we only need to support 14.0V, but it's easier to do the math with 15.0V.  Realistically we don't need to support dynamic faults when the alternator is running overvoltage, just installation errors."

rusefi, 2020-02-25:
Sweet, so we go forward with removal of

43;"D10,D26,D25,D24,D18,D12,D11,D9,D8,D7,D6,D5,D4,D3";"SOT-23";14;"BAS70-04";;;

jharvey, 2020-04-05:
This is obsolete, suggest closing this issue

closed 2020-04-05 by rusefi