rusefi/hw_microRusEfi #136: Research what else could be JLCPCB-assembled

rusefi, 2020-02-25:
Are there any https://lcsc.com/ components to use in place of
D2?
Q1?
etc everything else red

[image]

rusefi, 2020-02-25:
@DonaldBecker says

"D2 is the optional TVS for the CAN bus.  It isn't really needed.  It might protect against a lightning strike, but it would just save that part of the chip as everything else blows up.

Q1 is the pre-regulator MOSFET.   We are using the one specified and characterized for the TLE8888.  Presumably any similar part would work, but we can't calculate the stability without access to the proprietary info about the feedback.  It's $1-ish and easy enough to solder, so it doesn't strike me as a good part to substitute."

mck1117, 2020-03-30:
We can install C133648 as D41.  It's technically optional for normal operation, but the board might explode without it in case of an alternator load dump (battery disconnected from alternator while charging).  A racecar kill switch can do this, for example.

rusefillc, 2020-07-31:
Looks like we are in a great spot with pre-assembly at this point

closed 2020-07-31 by rusefillc