rusefi/hw_microRusEfi #137: Functional test board PCB

rusefi, 2020-02-27:
it would be nice to replace hand-made test board with a proper PCB

0) the board should accept wires from 48 pin pigtail
1) the board would need a few LEDs to indicate 5v & 12v
2) the board should have LEDs to test all MRE outputs
3) the board should have buttons to test CAM inputs
4) the board should have some resistor dividers to test analog inputs

rusefi, 2020-02-27:
Current state of this (shown one of two boards, complete test jig takes two boards at the moment, each board with 24 wires)

[image]

rusefi, 2020-03-07:
See also #140

rusefillc, 2021-02-07:
done

closed 2021-02-07 by rusefillc