rusefi/hw_microRusEfi #139: 0.4.8 JLCPCB DFM Report

rusefi, 2020-03-02:
C11 | Yes | C53987 | Guangdong Fenghua Advanced Tech 0603B472K500NT | 0603 | Cannot find this Pad in your PCB
-- | -- | -- | -- | -- | --
R121 | Yes | C17901 | Uniroyal Elec 1206W4F1000T5E | 1206 | Cannot find this Pad in your PCB
R123 | Yes | C17901 | Uniroyal Elec 1206W4F1000T5E | 1206 | Cannot find this Pad in your PCB
D27 | Yes | C2296 | Hubei KENTO Elec 17-21SUYC/TR8 | LED_0805 | Cannot find this Pad in your PCB
FB1 | Yes | C1034 | Sunlord SDFL1608Q4R7KTF | 0603 | Cannot find this Pad in your PCB
R125 | Yes | C17901 | Uniroyal Elec 1206W4F1000T5E | 1206 | Cannot find this Pad in your PCB
C41 | Yes | C23630 | Samsung Electro-Mechanics CL10A225KO8NNNC | 0603 | Cannot find this Pad in your PCB
U1 | Yes | C12345 | STMicroelectronics STM32F407VGT6 | LQFP-100_14.0x14.0x0.5P | Cannot find this Pad in your PCB
R34 | Yes | C23178 | Uniroyal Elec 0603WAF4703T5E | 0603 | Cannot find this Pad in your PCB
R77 | Yes | C25804 | Uniroyal Elec 0603WAF1002T5E | 0603 | Cannot find this Pad in your PCB
R62 | Yes | C25804 | Uniroyal Elec 0603WAF1002T5E | 0603 | Cannot find this Pad in your PCB
R63 | Yes | C25804 | Uniroyal Elec 0603WAF1002T5E | 0603 | Cannot find this Pad in your PCB
R76 | Yes | C25804 | Uniroyal Elec 0603WAF1002T5E | 0603 | Cannot find this Pad in your PCB
C2 | Yes | C134515 | Lelon VZH101M1HTR-0810L | SMD-ECAP-8x10 | Pad to Pad Spacing is too large
RN1 | Yes | C20197 | Uniroyal Elec 4D03WGJ0102T5E | 0603_x4 | Soldering area is too small
RN2 | Yes | C29718 | Uniroyal Elec 4D03WGJ0103T5E | 0603_x4 | Soldering area is too small
RN13 | Yes | C29718 | Uniroyal Elec 4D03WGJ0103T5E | 0603_x4 | Soldering area is too small
RN4 | Yes | C29718 | Uniroyal Elec 4D03WGJ0103T5E | 0603_x4 | Soldering area is too small
RN6 | Yes | C29718 | Uniroyal Elec 4D03WGJ0103T5E | 0603_x4 | Soldering area is too small
RN3 | Yes | C29718 | Uniroyal Elec 4D03WGJ0103T5E | 0603_x4 | Soldering area is too small
RN5 | Yes | C29718 | Uniroyal Elec 4D03WGJ0103T5E | 0603_x4 | Soldering area is too small

rusefi, 2020-03-02:
[image]

[image]

rusefi, 2020-03-02:
Simon says D16 rotated 180 deg
Simon says D19 gets a little wider

rusefi, 2020-03-03:
Just checked with 0.4.7 - yes, D16 needs to be flipped. Correct file is how the boards I have look like.

cocamousse, 2020-03-05:
Side note on C2, with a slight board change from the original 8*10.5 footprint to a 8*10 footprint you can use JLCPCB components. Guessing you know, just pointing it out.

rusefi, 2020-03-05:
@cocamousse please note how these pictures show C2 in black and for example SW2 not black

C2 in black means we already use C2 from JLCPCB components :)

cocamousse, 2020-03-05:
ah, sorry about that, I was looking at the report in the first post.

rusefi, 2020-05-24:
Seems like not an issue now

closed 2020-05-24 by rusefi