rusefi/hw_microRusEfi #140: Functional test board PCB - without pigtail

rusefi, 2020-03-07:


rusefi, 2020-03-07:
Need to find larger pogo pins - current pins are smaller than connector holes on microRusEfi

[image]

rusefi, 2020-03-07:
LED board would use same footprint as real MRE board

the idea is for LED board to use #6 bolts in order to sit on the board being test - #6 bolts match the mounting holes perfectly

MRE would stimulate itself and output low-side injector and high-side ignition signals. LED board should display those outputs using LEDs

[image]

[image]

rusefi, 2020-03-10:
P100-E2 pogo pins seem relevant

rusefi, 2020-05-11:
There is a plan to make progress at https://github.com/rusefi/hw_microRusEfi/tree/master/test_fixture

@jharvey is on the job!

rusefillc, 2021-02-07:
done

closed 2021-02-07 by rusefillc