rusefi/hw_microRusEfi #154: External +5v power source

rusefi, 2020-04-09:
At the moment we have USB on the 48 pin but we cannot feed +5v into MRE via the connector

this is forcing users to power from +12v which is not safe for firmware flashing

is there any chance to allow external +5v into MRE?

dron0gus, 2020-04-09:
Use one of analog inputs to feed USB 5V? Through diode to avoid influence during normal operation? Diode can also be used as jumper - it can be removed if user wants analog functionality only.

rusefi, 2020-04-12:
how about AV10?
1. i like that it's on the edge pin on the connector
2. I like "10" as in "often not used"

rusefi, 2020-05-24:
we have this on 0.5.0

so far I believe this was NOT tested

rusefi, 2020-07-10:
looks like this works

closed 2020-07-10 by rusefi