rusefi/hw_microRusEfi #157: RED FATAL/CRITICAL LED is a bit dim when it should be OFF

rusefi, 2020-05-11:
MRE has 4 status LED

somehow WARNING and RUN are properly totally OFF while FATAL/CRITICAL is a bit dim while should also be completely OFF

is this HW or SW issue?

dron0gus, 2020-05-11:
At early init code all pins are set to default mode (input mode with pull-up enabled).
Pull-up current is enough to light LED a bit.
Later some LEDs gpios reconfigured to output mode. Red led is not used and left as is.
Same thing cause all injectors channels to be activated for a few hundred mS during board boot/reset. This issue is HW fixed for 0.5.0 and need SW fix for <= 0.4.9

rusefi, 2020-05-11:
But we have
```#if EFI_PROD_CODE
	enginePins.errorLedPin.initPin("led: ERROR status", LED_ERROR_BRAIN_PIN);
	errorLedPort = getHwPort("primary", LED_ERROR_BRAIN_PIN);
	errorLedPin = getHwPin("primary", LED_ERROR_BRAIN_PIN);
#endif /* EFI_PROD_CODE */
```

is not that addressing the issue?

just to confirm - we are talking about long-term dim, not just quick start-up dim

dron0gus, 2020-05-11:
```
grep LED_ERROR_BRAIN_PIN -R ./config/
./config/boards/skeleton/efifeatures.h:#ifndef LED_ERROR_BRAIN_PIN
./config/boards/skeleton/efifeatures.h:#define LED_ERROR_BRAIN_PIN GPIOD_14
./config/boards/kinetis/efifeatures.h:#define LED_ERROR_BRAIN_PIN GPIOD_14
./config/boards/prometheus/efifeatures.h:#undef LED_ERROR_BRAIN_PIN
./config/boards/prometheus/efifeatures.h:#define LED_ERROR_BRAIN_PIN GPIOA_13
./config/stm32f4ems/efifeatures.h:#ifndef LED_ERROR_BRAIN_PIN
./config/stm32f4ems/efifeatures.h:#define LED_ERROR_BRAIN_PIN GPIOD_14
```
None of these is GPIOE_3 as it should be for MRE.
But:
```
dron@DronG:/home/work/rusefi/rusefi/firmware$ grep GPIOE_3 -R ./config/
./config/boards/microrusefi/board_configuration.cpp:	//engineConfiguration->atalErrorPin = GPIOE_3;		// d21 = red
./config/boards/microrusefi/board.mk:  EFI_FATAL_ERROR_PIN = -DEFI_FATAL_ERROR_PIN=GPIOE_3
```
Unfortunately EFI_FATAL_ERROR_PIN is not used anywhere in code.

rusefi, 2020-05-11:
Wow great find @dron0gus ! let me fix that

closed 2020-05-11 by rusefi