rusefi/hw_microRusEfi #160: text fixtures: buttons should have through-hole option or same as MRE

rusefi, 2020-05-13:
at the moment text fixture uses some unique button footprint

let's share footprint with MRE or better allow through-hole option. I suspect that my through-hole push buttons are 0.2x0.2 inch pattern

jharvey, 2020-05-13:
??? thru hole PB ???

rusefi, 2020-05-14:
[image]
@

rusefi, 2020-05-14:
that's the button from rusEFI 100 pin brain board

jharvey, 2020-05-14:
I see schematic for TL1105 found here https://sten-eswitch-13110800-production.s3.amazonaws.com/system/asset/product_line/data_sheet/144/TL1105.pdf

I see KICAD footprint Button_Switch_THT:SW_TH_Tactile_Omron_B3F-10xx appears to be proper. But I do not see a symbol for it that matches the above. It looks like a symbol would need to be created.

rusefi, 2020-05-14:
https://github.com/rusefi/rusefi/tree/master/hardware/brain_board is rusEFI hardware which uses that exact button

rusefi, 2020-05-14:
this is less important since JLC will mount SMD toggle switches for us