rusefi/hw_microRusEfi #165: Maybe go back to original USB connector

rusefi, 2020-05-15:
0.1 to 0.4.6 or 0.4.7 used one connector

with 0.4.7 @DonaldBecker has switched part number see https://github.com/rusefi/hw_microRusEfi/commit/8ef83eb0126df593adba10446c40f990759f1c9b

rusefi, 2020-05-15:
the complaint about Wuerth is 1mm lead length i.e. harder to solder into 1.6 pcb

and at some point Wuerth was not available on Moused which has triggered us to look at Digikey which suggested GCT as "similar option" and we got hooked on 1.6mm lead length.

rusefi, 2020-05-15:
Maybe we should even switch to mini USB since these are sturdier for vertical mount?

rusefi, 2020-05-17:
Two votes against USB-C due to soldering complexity.

rusefi, 2020-05-17:
@mck1117 points at https://lcsc.com/product-detail/USB-Connectors_Jing-Extension-of-the-Electronic-Co-Jing-Extension-of-the-Electronic-Co-TYPE-CFemale-14PIN-DIP_C168704.html  C168704

rusefi, 2020-06-20:
https://rusefi.com/forum/viewtopic.php?f=4&t=1730 was a failure of GCT

[image]

[image]

rusefi, 2020-06-21:
Would Lcsc mini usb Part C46396 fit existing footprint?

rusefi, 2020-07-06:
random eBay vertical mini USB - footprint is similar style to what we use but larger spacing between pins

[image]

jharvey, 2020-07-06:
Can we find a digikey equiv of that thru hole USBC connector? I like USBC as it's more robust and it's not "obsolete". If not then can we find a USB Mini from both Digikey and cheap alternative like ebay or LSCS.

rusefi, 2020-07-06:
@jharvey 732-8218-1-ND looks very very different from out current style

if it would be a 24 pin connector I would be worried about soldering

at this point going back to mini seems to be the safest option

rusefi, 2020-07-06:
How about UE25BE55100 ? I wonder if the pinout is similar in terms of which pin is located where

rusefi, 2020-07-06:
@jharvey what about Frankenso vertical pinout?

I believe Frankenso vertical Mini USB has pins in the same positions

footprint does not match, we either need to make holes a bit larger or just switch

[image]

[image]

jharvey, 2020-07-07:
How about that universal one we once were developing? It would look like this. 
[image]

stefanst, 2020-07-12:
Vote against vertical mount Micro-USB. Mine has failed after frequent use. There was, however, no undue ABUSE. Just plugging, unplugging a few hundred times. Each time likely bending the connector slightly.  Any connector without WIDE solder tabs will fail after a time, especially if installed vertically which exacerbates bending moments.
[image]
[image]

rusefi, 2020-07-12:
[image]

rusefi, 2020-07-12:
it's GCT which has failed

[image]

rusefillc, 2020-07-21:
USB-C option:
Dimitri C says "LCSC C168704
It's actually 14 pins now that i looked at it but fairly easy to solder.

It's 7 pins X2 for each side"

rusefillc, 2020-07-24:
0.5.2 with miniUSB 

[image]

[image]