rusefi/hw_microRusEfi #171: VR input: R9 and R15 useful or not?

rusefi, 2020-05-20:
R9=R15 DNP = more sensitive

both R9 and R15 in place = lower signal level, possibly necessary for extremely hot sensors

Matthew Kennedy  says: 
I have no data to support that they are required since the tle8888 has internal clamp diodes to prevent overvoltage
so I might even go so far as to say that we should just remove them
the tle8888 datasheet application diagrams don't mention having a resistor there
I have no idea why we have them honestly

jharvey, 2020-05-23:
Perhaps change for TVS diodes, that kick in when above a specific voltage, then they don't attenuate when low RPM. Very similar to the 8888's internal TVS. The "VR Sensor Interface Input Clamping Voltage" is as low as 2V, and "VR Sensor Interface Input Clamping Current" is a max of 50mA. I see 3.3V TVS are available from JLC, perhaps we can find a lower voltage diode. https://datasheet.lcsc.com/szlcsc/ProTek-Devices-PSD03-LF-T7_C139297.pdf

rusefi, 2020-06-08:
let's just DNP R9 and R15?

rusefi, 2020-06-13:
it was decided not to populate R9 and R15

see also #178

rusefillc, 2020-07-27:
0.5.2 was fabricated without R15 but with R9

i think that's similar to the R2 issue, we need to DNP R9 a bit more :) @jharvey 

[image]

Qwerty-OFF, 2022-04-23:
[image]

closed 2022-04-23 by Qwerty-OFF