rusefi/hw_microRusEfi #172: test board I had: wire buttons to the analog inputs

rusefi, 2020-05-21:
@mck1117 says "one idea I had for a test board I had: wire buttons to the analog inputs, and LEDs to outputs.  Write firmware that just reads inputs and writes them to each output,  That way you can just press buttons and make sure exactly one LED turns on."

we have the outputs. We have JLC soldering switches for us. I think that's a great idea!!!

mck1117, 2020-05-21:
Another more fully fleshed-out option for this is to have a second MCU and CAN interface on the test board.  The ECU-under-test can accept commands to toggle outputs over CAN, whole broadcasting the state of the inputs.  Test board MCU applies stimulus (analog input voltage, digital inputs, crank/cam signal), and checks that CAN reports the right thing.  Then the test MCU can toggle each output over CAN, checking that the pin changes as expected.  That way we could just turn on a single green light that says "board ok" instead of having to check every output manually.

mck1117, 2020-05-21:
(but that's of course a bunch more work, and only a required thing at scale)

jharvey, 2020-05-21:
I tend to think pre-programmed chip like the one found on the discovery. Let is automagically smash a program into MRE and do some tests.

jharvey, 2020-05-23:
Why not connect outputs, to inputs and let the MRE push the buttons? Similar to AV6, AV8, AV9, and AV10.