rusefi/hw_microRusEfi #175: Mark R2 DNP

rusefi, 2020-06-12:
R2 watchdog connect is preventing stm32 from start unless we have tle8888 powered via +12v

let's DNP R2 for now

rusefi, 2020-07-10:
@jharvey could this be closed?

jharvey, 2020-07-22:
Hmmmm, I see 
[image]
However the BOM comes out like this. 
[image]

reportingsjr, 2020-07-24:
This is an issue with your BOM script, not with KiCad.
If you look at the BOM script it is grouping components only by footprint type and value, hence why it is grouping your DNP resistor in there.

Change the R2 value to DNP and I bet it will separate it out.

I do wish KiCad had some other default fields like DNP so scripts could be more consistent on this!

jharvey, 2020-07-27:
Made "0R" into "0R DNP" This now should work long term.

closed 2020-07-27 by jharvey

MHTSOSVOODOO, 2020-07-29:
We could replace R2 with a diode with the cathode pointing towards the reset button. This way you can still reset TLE8888 when you press the button but it can't pull the reset line to ground when 12V is missing. I tried it and it seems to work.