rusefi/hw_microRusEfi #178: Hall input: do we need R18=R19=4.7K? keep C12 & C35?

rusefi, 2020-06-13:
Would TLE8888 in Hall mode allow us to drop R18=R19 altogether?

In order to reduce HW difference between hall and VR modes can we keep C12 & C35 while running in Hall mode?

rusefi, 2020-06-17:
My NB2 miata has just started without R18 and R19

stwillocks, 2022-04-23:
I have 2 seperate MRE with 202204 pre release firmware

MRE1=R9,C12,C35,R15 NOT populated & R19,R17,R18 populated 
MRE2=R9,C12, C35, R17,R18,R19 populated

MRE2 works but sens trigger error at certain RPMs. 
At 2k-2.5k trigger error but it has continuity 
At 4k RPM with low throttle it stucks at 4k but with full throttle it gets over 4k but stucks 6.4k RPM

MRE1 revs up all the way to 6.4k but while deceleration it gives trg error at 4k. and mixing signals

rusefillc, 2022-04-23:
MRE2 with all populated is pretty strange see https://github.com/rusefi/rusefi/wiki/Hardware-microRusEFI-kit-instructions

stwillocks, 2022-04-23:
I have slightly wiggle at crank trigger wheel, which i discovered recently, I believe it caused "~4k trg error"s. Monday I am going to get the wheel machined and balanced then try different combinations, will post logs & results.  BTW thanks for notation.

rusefillc, 2022-04-23:
#223 says hi

[image]

rusefillc, 2025-08-05:
See also https://github.com/rusefi/rusefi/wiki/Hardware-microRusEFI-kit-instructions#crankshaft-position-sensor-options-hall-or-vr

rusefillc, 2025-08-05:
We are good with https://github.com/rusefi/hw_microRusEfi/issues/236 BOM

closed 2025-08-05 by rusefillc