rusefi/hw_microRusEfi #180: pin contention - TLE8888 + CS + BT

rusefi, 2020-06-22:
TL,DR: do we have ANOTHER set of UART pins exposed on MRE not PC10&PC11?

How do we get MRE to
1) obviously keep using SPI for TLE8888
2) have different SPI for SD card
3) and also have UART for Bluetooth?

TLE8888 uses TBD pins which conveniently could be multiplexed to SPI1 or SPI3
SD card ideally should be running on a different SPI to reduce contention. We have complete SPI exposed which uses PC10&PC11

so if PC10 and PC11 is used by SPI for SD, do we have another pair of pins exposed on all the little random vias? if not, what would be the minimal HW change to have uart exposed without sharing pins with SPI?

rusefi, 2020-06-22:
@ksmola since you did UART recently poke :)

rusefi, 2020-06-22:
see table on the bottom of https://github.com/rusefi/rusefi/wiki/Hardware_microRusEfi_wiring

rusefi, 2020-06-24:
[image]

rusefi, 2020-06-24:
[image]

how about PB10 & PB11 which are used for SPI Flash right now? Does SPI use them simply as GPIO?

jharvey, 2020-07-06:
I'm not following the request. Is this correct?
PB10 on SD card --> PE0
PB11 on SD card --> PC13
Then we want 
PB10 to a via for UART via soldered wire
PB11 to a via for UART via soldered wire

rusefi, 2020-07-06:
The request:
before the change PB10 PB11 were routed to SPI Flash (not SD card)
PE0 PC13 were available to outside world via VIAs

The request:
swap pin usage. use PE0 PC13 to control SPI Flash
make PB10 PB11 available to outside world via VIAs

@jharvey

jharvey, 2020-07-06:
My key concern is if PE0 or PC13 should be connected to the flash traced called HOLD? I have committed a draft copy. Did I get the correct signal to the correct pin? If they are flipped does that matter? Is the PCB as it's drawn now correct and ready for MFG files?

rusefi, 2020-07-06:
Does not matter which one is which, it's GPIO so full programmatic control

jharvey, 2020-07-06:
Thanks for the verification. This is routed, so we can close it.

closed 2020-07-06 by jharvey

rusefi, 2020-07-14:
Before the change, 0.5.1 and older

[image]