rusefi/hw_microRusEfi #183: [0.5.2] D41 orientation seems wrong in

rusefi, 2020-07-10:
[image]

[image]

[image]

jharvey, 2020-07-12:
How do we make the rotation script into a submodule? I think the issue is that we did not update the multiple projects to have the same specified rotation file.

rusefillc, 2020-07-26:
I am confused. I've rotated D41 in placement file and submitted a new order, and it's STILL rotated?!

[micro_rusEFI_JLC-CPL_R0.5.2_from_order.csv.txt]
(https://github.com/rusefi/hw_microRusEfi/files/4977757/micro_rusEFI_JLC-CPL_R0.5.2_from_order.csv.txt)

[image]

[image]

jharvey, 2020-08-08:
Rotation updated, closed.

closed 2020-08-08 by jharvey

rusefillc, 2020-08-11:
@jharvey D41 history
```
./Archived/rusEFI-Micro_R0.4.7-cpl.csv:D41,76.835000mm,-98.425000mm,Top,90
./0.4.8/rusEFI-Micro_R0.4.8-cpl.csv:D41,76.835000mm,-98.425000mm,Top,90
./0.5.0/rusEFI-Micro_R0.5.0_JLC_cpl.csv:D41,76.835000mm,-98.425000mm,Top,90
./R0.5.1/microRusEFI_R0.5.1_JLC-CPL.csv:D41,76.835000mm,-98.425000mm,Top,270
./R0.5.2/micro_rusEFI_JLC-CPL_R0.5.2.csv:D41,76.835000mm,-98.425000mm,Top,90
./R0.5.3/micro_rusEFI_R0.5.3_JLC_CPL.csv:D41,SM15T33C,76.835000,-98.425000,270.000000,top

```