rusefi/hw_microRusEfi #197: Resistors in between header pins are a terrible PITA

stefanst, 2020-09-04:
[image]
[image]
Tiny little components placed between header pins are terrible for hand-soldering the pins.
35 was connected to R54 and R52 (GND). Fortunately it's a low-side output, so short to GND didn't kill anything. I hope nothing died on any of the other boards. At least I didn't smell anything.
The gap is just way to small for something that's supposed to be hand-soldered.
I understand that this solution is clever, but it kills assembly of the boards. I have to go over each board of the 5 assembled now with a microscope and see where they're bridging. And then try to remove the bridges very carefully with thin braid and a skinny tip.
I'm sure we can find some other space on this board to place those resistors. There should be no components in between those header pins.

JRDMcLAREN, 2021-12-27:
Can I try to do something with this .?
Is any space on bottom side on ALU case .? (asume yes)

JRDMcLAREN, 2022-01-03:
Done in my RC2 - https://github.com/JRDMcLAREN/hw_microRusEfi https://rusefi.com/forum/viewtopic.php?p=43789#p43789