rusefi/hw_microRusEfi #203: 6-11 cylinder flow bench test

rusefillc, 2020-11-14:
MRE has TLE8888+2xVNLD5090

TLE has 6 high-current outputs but those are 4x550 mOhm + 2 or 3 x of 350 mOhm outputs.

2xVNLD5090 gives us 4x180 mOhm outputs.

So while theoretically that's 10 channels for injectors, we do not have 6 channels which would use same kind of transistors.

Open question if 6, 8 or 10 injectors on a flow bench would behave same or differently. Only way to tell is to flow those :)

rusefillc, 2020-11-14:
PS: need to run on 80% duty cycle

nmschulte, 2022-10-18:
I'd love to help tackle an answer to this -- maybe next year when my project is running in stock form again, and I have a rusEFI compatibly hardware in-hand.

rusefillc, 2022-11-14:
PS: either way MRE would really run out of outputs really fast in 5+ cylinder application, a larger ECU like Proteus or 8chan is needed for 5+ cylinders

rusefillc, 2022-12-19:
v8 see https://rusefi.com/forum/viewtopic.php?p=46845#p46845