rusefi/hw_microRusEfi #212: Chip Shortage: TLE8888 options

rusefillc, 2021-12-03:
https://octopart.com/search?q=TLE8888

[image]

rusefillc, 2021-12-25:
[image]

rusefillc, 2022-01-20:
No changes

[image]

rusefillc, 2022-08-09:
@JRDMcLAREN are you still getting TLE8888 anywhere or do you just have a lot remaining?

JRDMcLAREN, 2022-08-09:
I have last 5 pcs. I'm bought "last" 18 pcs in January 2022 from Digikey
..and have last 3 pcs F407VGT6 ....

rusefillc, 2022-12-11:
@JRDMcLAREN moused has stock 

[image]

JRDMcLAREN, 2022-12-12:
THX, 5pcs ordered ..
and what kind of MCU can be used in near future ... F427VGT7 .. maybe ..
[image]

rusefillc, 2022-12-12:
427vg is already used by MRE

JRDMcLAREN, 2023-01-19:
TLE8888QKXUMA1 is marked as "End Of Life" on Mouser
[image]

rusefillc, 2023-01-20:
it's a bit more complicated - primary version is active

[image]

JRDMcLAREN, 2023-02-05:
email form Mouser 26/01/2023
[image]

rusefillc, 2023-02-06:
@JRDMcLAREN please clarify what are you saying.

@JRDMcLAREN are you saying that mouser info is more reliable than https://www.infineon.com/cms/en/product/power/automotive-conventional-powertrain-ics/engine-management-ic/tle8888qk/ official status which shows "active" for me today?

JRDMcLAREN, 2023-02-06:
My experience is, when Mouser has some product marked as EoL, then it is END OF LIFE.
..we need to make "new MRE" with any other chip, maybe ....

YES, Infineon says ACTIVE, but datasheet date is 05/2017, Product Brief is 08/2015 and Footprint is 02/2019

..just try to think about it...

(I have last 10pcs on stock)

mi-hol, 2023-02-18:
> when Mouser has some product marked as EoL, then it is END OF LIFE.

I fully agree with this statement. This would mean the end for producing new MRE boards with current design & components.
What options for replacing TLE8888 do we have?

rusefillc, 2023-02-18:
I've just ordered enough chips that I can worry about it all later

[image]

mi-hol, 2023-02-18:
Or is maybe [EVAL-L9779WD-SPI](https://www.st.com/resource/en/data_brief/eval-l9779wd-spi.pdf) Evaluation board for L9779WD-SPI an alternative for the future?

Features
- Voltage min/max: 4.15V to 28V.
- No. channels:
  * 4 injector driver
  * 1 canister valve driver
  * 2 Lambda sensor heater
  * 1 Main relay driver
  * 6 relay driver
  * 1 LED driver
  * 4 Ignition driver
- A programmable power stage configurable as: 
double H-bridge to drive a stepper motor driver,
or 
as independent 4 High side and 4 Low side drivers.
- Device registers setting and the full diagnostic are available through SPI.
- Communication protocols: SPI, CAN, (both with dedicated connectors), Kline.
- Access to all relevant pins by test points. Input signal connector compatible with the SPC56MDiscovery
- Possibility to **connect a generic microcontroller boards by using a simple adapter**.

rusefillc, 2023-02-18:
Let's move this to https://rusefi.com/forum/ where that amazing chip is already discussed in detail

rusefillc, 2023-04-01:
[image]

rusefillc, 2023-04-01:
[image]

rusefillc, 2023-04-18:
[image]