rusefi/hw_microRusEfi #214: Diff of microRusEFI with SD card

rusefillc, 2022-04-01:
Looking to have a diff between https://github.com/rusefi/hw_microRusEfi master and https://github.com/rusefi/hw_microRusEfi/pull/211

rusefillc, 2022-04-01:
@chuckwagoncomputing any chance you could help?

@Qwerty-OFF FYI

chuckwagoncomputing, 2022-04-01:
Here's your diff. This is a diff to the point from which @JRDMcLAREN made his fork, not from the current master.

[diff.tar.gz](https://github.com/rusefi/hw_microRusEfi/files/8401127/diff.tar.gz)

JRDMcLAREN, 2022-04-02:
I have some unpopulated PCBs. Can send it to somebody, if it helps.

rusefillc, 2022-04-09:
@Qwerty-OFF  - un-compress https://github.com/rusefi/hw_microRusEfi/issues/214#issuecomment-1086379574 folder  and open web index.html in browser

closed 2022-04-09 by rusefillc