rusefi/hw_microRusEfi #216: Fab options

rusefillc, 2022-04-08:
1. The SM15T33CA on D41, it's out of stock, could we order the SM15T33CAY?
2. The IPD30N06S2L23ATMA3 on Q1, it's out of stock, the price is quite high on market, do you have an alternative part to recommend?

rusefillc, 2022-04-08:
@Qwerty-OFF need your help

@jharvey :) do you remember about MRE? because MRE remembers about you.

rusefillc, 2022-04-08:
2. IRLR2905TRPBF
JLCPCB Part C3010
TO-252-2
55V 42A 110W 27mΩ@10V,25A 2V@250μA N Channel TO-252-2(DPAK) MOSFETs ROHS

Qwerty-OFF, 2022-04-08:
1. Yes

rusefillc, 2022-04-09:
2. what about [IRLR2905ZTRPBF](https://octopart.com/irlr2905ztrpbf-infineon-65875016?r=sp) or [AUIRLR2905](https://octopart.com/auirlr2905-infineon-71801451?r=sp) or AUIRLR2905Z?

rusefillc, 2022-04-09:
2. @mck1117 says AUIRLR2905 looks like closes thing to IRLR2905TRPBF
Z options are probably fine as well

rusefillc, 2022-04-09:
jlcpbc has Z option C68209

Qwerty-OFF, 2022-04-09:
Lower resisance (good), higher gate threshold (non-critical)
IRLR2905TRPBF:
[image]
IRLR2905ZTRPBF:
[image]
Other characteristics are similar, better, or not important to us

closed 2022-04-09 by rusefillc